lilka-dev/lilka #128: APP_FLAG_FULLSCREEN ignored by panel

noisedsn, 2024-12-13:
Не зникає панель під час роботи додатків, в яких встановлено APP_FLAG_FULLSCREEN та НЕ встановлено APP_FLAG_INTERLACED, таких як NesApp, ModPlayerApp, AbstractLuaRunnerApp та інші. В результаті, вгорі екрана відображається панель, а додаток рендериться зміщеним на 24 пікселі вниз і верхні рядки 24...48 займають артефакти від накладання нижньої частини кадру додатка на верхню.

[image]

Відчуваю, що це, можливо, якось пов'язано зі змінами щодо інтерлейсу. Якщо примусово малювати все через інтерлейс, то баг зникає в усіх додатках. Для цього я спробував замінити в файлі appmanager.cpp
                    lilka::display.drawCanvas(app->backCanvas);
на 
                    lilka::display.drawCanvasInterlaced(app->backCanvas, app->frame % 2);

На жаль, докопатися до суті і пофіксити коректно моїх скромних знань не вистачило...

frostmorn, 2024-12-13:
can't reproduce

[image]
[image]
[image]
[image]
[image]
[image]
[image]

noisedsn, 2024-12-13:
> can't reproduce

А скріншоти і у мене красиві. У фреймбуфері дані коректні. здається, а на екрані мерехтять 24-48 рядки і панель стирчить у повноекранних додатках (крім лілтрекера, до речі). Може якісь таймінги в залізі, чи глюки компілятора - хтозна, я не справжній програміст. Поки що вдалося некрасиво пофіксити, додавши у display.cpp ще одну функцію аналогічну тій що відмальовує інтерлейс:

void Display::drawCanvasBad(Canvas* canvas) {
    this->startWrite();
    for (int y = 0; y < canvas->height(); y ++ ) {
        this->writeAddrWindow(canvas->x(), canvas->y() + y, canvas->width(), 1);
        this->writePixels(canvas->getFramebuffer() + y * canvas->width(), canvas->width());
    }
    this->endWrite();
}

ну і в display.h:
    void drawCanvasBad(Canvas* canvas);

Після чого в appmanager.cpp:
lilka::display.drawCanvas(app->backCanvas);
на
lilka::display.drawCanvasBad(app->backCanvas);

noisedsn, 2024-12-14:
Тим не менш, зі штатною функцією глюк є, з моїм костилем - немає. Не одразу помітив, що панель не відображається як звичайно, то просто "слід" від неї, на якому не йде годинник і нічого не міняється.

https://github.com/user-attachments/assets/2ec24964-374b-4f5c-b61f-900b69038e58

Таке враження наче десь якась команда на зміну координат губиться, а моя тормозна функція трохи все сповільнила. Пробував потикати в різні місця delay - не допомагає.  Замовив другий дисплей, приїде - спробую з ним, може мені глючний попався.

noisedsn, 2024-12-19:
Мені дико соромно, в мене був не найновіший зріз репозиторію. В останній прошивці все коректно.

closed 2024-12-19 by noisedsn